aritchie/bluetoothle #381: [Bug]

frankhommers, 2020-04-14:
# IF YOU DON'T ANSWER THIS TEMPLATE - THE BOT WILL AUTOMATICALLY CLOSE YOUR ISSUE!

## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [ ] iOS
 - [ ] Android
 - [ ] UWP
 - [X] .NET Standard

## Version of Library

6.3.0.19

## Version of OS(s) listed above with issue

Linux raspberrypi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux

## Steps to Reproduce
1. The list of supported platforms differs on the README.md here and https://www.nuget.org/packages/Plugin.BluetoothLE/, so that needs fixing.
2. I get confused when I try to run it it shows this message: "[Plugin.BluetoothLE] No platform plugin found.  Did you install the nuget package in your app project as well?"
3. When I try to build the solution myself, I see the NETStandard stuff commented out for compilation in the .csproj

## Expected Behavior

To be able to run it using Linux and Bluez.

## Actual Behavior

Does not work.

aritchie, 2020-04-14:
Linux and bluez were never supported by this library.  There is some play in the codebase, but that is all

closed 2020-04-14 by aritchie

frankhommers, 2020-04-16:
Then remove it ".NET Standard" from https://www.nuget.org/packages/Plugin.BluetoothLE/ please.

aritchie, 2020-04-16:
@frankhommers Please be respectful - this isn't your library.  .NET standard is required for this library to work properly in the Xamarin/UWP space

frankhommers, 2020-04-17:
I think I am respectful. 

I just wasted a few hours on this trying to run it on .NET Standard with .NET Core on Linux and I was getting weird errors. I just want to save time of other people trying the same.