aritchie/bluetoothle #382: [Bug] Is there any way to make disconnect work under UWP?

kamilmode, 2020-04-21:
# IF YOU DON'T ANSWER THIS TEMPLATE - THE BOT WILL AUTOMATICALLY CLOSE YOUR ISSUE!

## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [ ] iOS
 - [ ] Android
 - [x ] UWP
 - [ ] .NET Standard

## Version of Library
6.3.0.19

## Version of OS(s) listed above with issue
Windows 10 IoT Core

## Steps to Reproduce
1. Scan for devices
2. Connect to device and use characteristics (read+write)
3. Disconnect

## Expected Behavior
Complete disconnect

## Actual Behavior
Device never disconnects - verified by bluetooth lookup not showing device on other devices. Is "Device cannot disconnect" problem still present under UWP? Any workarounds? The library is great but this is blocker for using it under UWP for me. Thanks for help!

aritchie, 2020-04-21:
If it's a blocker for you, you are welcome to contribute a fix.

kamilmode, 2020-04-21:
I might - could you please just tell me whether this is still an issue (UWP not disconnecting fully from devices)? Thanks